micropython/micropython-lib #181: urllib.parse: MemoryError: memory allocation failed, allocating 560 bytes

SlowBro904, 2017-05-20:
```
>>> import urllib.parse
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 560 bytes
>>>
```
Best I can tell it's generating the error on line 475 where _hextobyte is created.

I have a WiPy 2.0, using the 1.6.13.b1 firmware.

SlowBro904, 2017-05-20:
Also fails if I do from urllib.parse import unquote, which is the specific module I'm seeking. There is no extra code loaded; nothing in boot.py or main.py.

I found another module written specifically for the WiPy 1.0 that does work on the 2.0.
https://github.com/lucien2k/wipy-urllib/blob/master/urllib.py

pfalcon, 2017-06-14:
"MemoryError: memory allocation failed, allocating 560 bytes" means that your target device doesn't have enough memory to use this module.

closed 2017-06-14 by pfalcon

SlowBro904, 2017-06-15:
Device is a WiPy 2.0, which has more memory (512K) than the official board of uPython, the pyboard (192k). You might not want to ignore this issue. I shared a workaround with working code.

dpgeorge, 2017-06-15:
Thanks @SlowBro904 for the link to the other urllib.

Note that WiPy 2.0 only has about 70k of heap available for the user (the rest is taken by the system), while the pyboard has about 100k of heap for the user, so that could be why it doesn't work on your board.  Also, the urllib in this repo was taken (mostly) from upstream CPython and is not optimised for MicroPython.