micropython/micropython-lib #182: upysh - additional functions

pacmac, 2017-05-23:
Not sure if you are interested, but I added some additional functions to my version of sh which I find very useful:

    import sys,os,gc
    
    class FORMAT():
      def __repr__(self):
        import flashbdev
        os.VfsFat.mkfs(flashbdev.bdev)    
      def __call__(self):
        return self.__repr__()
        
    class REBOOT:
      def __repr__(self):
        import machine
        machine.reset()
      def __call__(self):
        return self.__repr__()
    
    class DF:
        def __repr__(self):
          root = sys.path[1].split('lib')[0]
          s = os.statvfs(root)
          print((s[0]*s[3]) / 1048576,'MB')
        def __call__(self):
            return self.__repr__()
    
    class FREE:
      def __repr__(self):
        F = gc.mem_free()
        A = gc.mem_alloc()
        T = F+A
        print('Ram Total:'+str(T)+' Free:'+str(F)+' ('+str('{0:.2f}'.format(F/T*100))+'%)')
    
      def __call__(self):
          return self.__repr__()

pfalcon, 2017-06-14:
The idea of upysh is to be small. Additional functions may be added one by one, but the code style must adhere to PEP8.

jonnor, 2024-08-25:
Not much has been added to this module over the last 5 years. In practice I think it is unmaintained. Proposing to close this issue. If anyone wants to improve it, it might be better done in a third-party repository. Which can easily be installed with `mip` these days.

closed 2024-08-27 by projectgus