micropython/micropython-lib #185: Password

scargill, 2017-05-31:
I am trying to use mqtt.simple (before adding mqtt.complex) on an ESP - I'm using Mosquitto and as always with username and password. For the life of me I cannot find an example of how to add the username and password.

Can someone please point me in the right direction.

dpgeorge, 2017-06-01:
You pass the username/password in the constructor: `MQTTClient('client', '192.168.0.123', user='username', password='mypassword')`

scargill, 2017-06-01:
Excellent - that works.  I've been reading around and the general conversation seems to indicate there is no re-connect/resubscribe mechanisms yet that work reliably - is that a fair comment?

dpgeorge, 2017-06-02:
> I've been reading around and the general conversation seems to indicate there is no re-connect/resubscribe mechanisms yet that work reliably

That's right, the robust implementation of umqtt doesn't do resubscription when it reconnects.

closed 2017-06-02 by dpgeorge