micropython/micropython-lib #189: socket: socket.socket is not fdfile compatible type

ThunderEX, 2017-06-18:
it seems socket.socket is different from usocket.socket, and it's not fdfile compatible type.

this leads to uselect.poll().register(`obj`, eventmask) can't take it as first argument `obj`.

pfalcon, 2017-06-21:
That's correct, builtin `usocket` module is compatible with builtin `uselect` module, and their integration is highly efficient (because involves only builtin, not user, types). micropython-lib's `socket` module would be compatible micropython-lib's with `select` module, except large part of their integration is not implemented, apparently because nobody needed it, as everyone prefers the efficient integration offered by "u" modules.

What are you trying to achieve?

jonnor, 2024-08-25:
Compatibility with CPython has been greatly improved over the last 5 years, including in the `socket` module. If there are specific issue, please re-open with details.

ThunderEX, 2024-08-26:
close as stale.

closed 2024-08-26 by ThunderEX