micropython/micropython-lib #190: Typing module

dbrgn, 2017-06-23:
Are there any plans to port the type annotations from the [typing module](https://docs.python.org/3/library/typing.html) to Micropython?

dpgeorge, 2017-06-26:
The MicroPython compiler itself supports type annotations, so in principle the typing module will work.  But there are no plans to port it.  Feel free to try doing it yourself (you can start just by copying the CPython implementation, like some of the other modules here) and reporting about it here or on https://forum.micropython.org

dbrgn, 2017-06-26:
Thanks. Which Python version - if any - does Micropython track?

dpgeorge, 2017-06-26:
> Which Python version - if any - does Micropython track

As per the README in the top level of the main repo (https://github.com/micropython/micropython/blob/master/README.md) it's Python 3.4 with a bit of extra stuff (eg async/await keywords) from version 3.5.

dbrgn, 2017-06-26:
Ah, thanks! I only looked inside the `micropython-lib` repo :)

So when porting, would it make sense to use the current 3.6 backport directly?

I could probably give it a stab this week. Would be useful to be able to typecheck micropython projects with mypy.

dpgeorge, 2017-06-26:
> So when porting, would it make sense to use the current 3.6 backport directly?

Yes, because that will likely work and will be more up to date (eg with bug fixes) than earlier versions.

dbrgn, 2017-06-26:
I just checked. The module makes a lot of use of the `abc` module which is not supported yet by micropython (and is probably not that easy to port).

It would probably be best to write dummy classes for all types. That way the typechecking could not be done on Micropython itself, but mypy running on regular python would probably work, with proper stubs.

pfalcon, 2017-06-26:
Few comments:

* Writing a module from scratch for micropython-lib is almost always a better choice than to port (CPython stdlib is very bloated).
* If taking from CPython, taking a version from older CPython release (3.3, 3.4) is almost always a better choice, because with each release, modules in CPython become more and more bloated.

dpgeorge, 2017-06-26:
> If taking from CPython, taking a version from older CPython release (3.3, 3.4) is almost always a better choice,

In this case, typing was introduced in 3.5 and has evolved a lot so 3.6 is probably a better implementation.

Cediddi, 2017-09-14:
I saw the micropython-typing module. I can provide a hand. I'm also writing stub files for esp8266 port (1.9.2 currently). I'll release it as soon as I finish machine. Stub files I'm writing are using typing library. It'd be great if micropython would expose a fake typing module. Just to prevent import errors. Two cases are important `from typing import something` or `import typing`. I don't think micropython should support `from typing import *` or give an error to say "you need to install micropython-typing, which would be a minimal typing library.

stlehmann, 2018-03-28:
I would really love to have a working typing module for Micropython. Sadly the CPython `typing` modules makes use of Metaclasses which are not supported by Micropython. I would prefer a lightweight `utyping` module that allows type-checking with Mypy over a fake module, though.

dbrgn, 2018-08-07:
@stlehmann yes, I agree (as already suggested in my [older comment](https://github.com/micropython/micropython-lib/issues/190#issuecomment-310976000)). Stubs would be sufficient, you don't want to do typechecking on the target platform.

stlehmann, 2018-08-08:
I added the fake modules in #297. It works for simple types like `Any`, `Callable`. But the generic types like `List[int]` or `Dict[str, int]` won' t work. This is because Micropython does not support the `__class_getitem__(cls, key)` special function that allows the creation of generic class objects.  @dpgeorge  would it be possible to add that feature?

jpochyla, 2018-08-15:
I think a better way would be to avoid running the typing code, rather than shoe-horning a static annotations into the runtime. For now, simple `if False:` works.

Cediddi, 2018-08-15:
I always use mpy-cross to compile py files before uploading to the board. Type annotations and stubs would help me develop my code faster, not run like a staticly typed program. Type annotations are ignored in runtime, they are just annotations.

jpochyla, 2018-08-15:
> Type annotations are ignored in runtime, they are just annotations.

Yes, but other typing code (like `typing` imports, newtypes, etc.) are not ignored. If no code is executed in the end, why would we need `__class_getitem__`?

Cediddi, 2018-08-15:
I'm not sure about __class_getitem__ 's necessity but I think mpy-cross can ignore that operations.

stlehmann, 2018-08-15:
> If no code is executed in the end, why would we need __class_getitem__?
I also thought no code will get executed if no type-checking is done. But if you use a type-annotation like this: 

    myobj: List[int]

in the interpreter you will get a SyntaxError on execution.

stlehmann, 2018-08-15:
Ok but with function annotations this works just fine:

    def foo(bar: List[int]) -> None:
        print("Hello")

This will execute without problems. Seems like type-annotations for simple objects is not supported by micropython.

jpochyla, 2018-08-15:
Yes, PEP 526 is not implemented in MicroPython.

stlehmann, 2018-08-15:
@Cediddi A typing module with fake typing classes leads to working micropython code with type-annotations included. There would be no need to compile it to get rid of the annotations anymore.

devxpy, 2019-10-07:
I wonder if it is possible to write a typescript like transpiler for micropython, so that an extra module ([~2000 lines](https://github.com/python/cpython/blob/master/Lib/typing.py)) isn't needed.

CallumJHays, 2020-12-28:
G'day - reviving this dead issue 🧟

> I wonder if it is possible to write a typescript like transpiler for micropython, so that an extra module ([~2000 lines](https://github.com/python/cpython/blob/master/Lib/typing.py)) isn't needed.

The dummy module provided by @stlehmann kind of gets the job done in significantly less code. It's not ideal, but I think a full type-annotation-removing transpiler would be overkill.

Anyway I'm commenting because @stlehmann 's pull request was only ever merged into **pfalcon**'s `micropython-lib` (now `pycopy-lib`). I can simply use [`pycopy-typing`](https://pypi.org/project/pycopy-typing/) but it would be nice if it could be mirrored in [`micropython-typing`](https://pypi.org/project/micropython-typing/) to remove confusion (not sure if there would be licensing issues here).

Also, some of my code utilizes the `typing_extensions` module for which there appears to be no equivalent for micropython. Would you accept a pull request with said module implemented (in the same fashion as @stlehmann's contribution)?

neelkarma, 2021-08-12:
Is it possible to at least have something like the `TYPE_CHECKING` property from the `typing` module so we can use autocompletion/typechecking in our code without the performance reduction and having to port over the entire typings library?

dlech, 2022-12-12:
> Is it possible to at least have something like the `TYPE_CHECKING` property from the `typing` module so we can use autocompletion/typechecking in our code without the performance reduction and having to port over the entire typings library?

How about something like this?

```python
try:
    from typing import TYPE_CHECKING
except ImportError:
    TYPE_CHECKING = False

if TYPE_CHECKING:
    from typing import ...
```

Or this?

```python
if getattr(globals(), '__annotations__'):
    from typing import ...
```

stinos, 2022-12-12:
@dlech hmm `TYPE_CHECKING` is one thing which I didn't include in the PR just made, any idea if this is commonly used?

dlech, 2022-12-12:
A quick github search for `TYPE_CHECKING` in Python files show 210k hits vs 3.1 million for `typing`, so not super-common, I guess. But useful when you want to avoid unnecessary imports at runtime.

andrewleech, 2022-12-12:
It's actually very important to have as it allows you to import things to satisfy type hinting that would otherwise cause circular dependencies. It's defined True by the type checking tools, should be False in the library

jonnor, 2024-08-25:
The https://github.com/micropython/micropython-lib/pull/297 merge request looks to have useful code. So if anyone is interested, they can take that code and re-base it on the latest master, so we can get it included.

Josverl, 2024-08-26:
@Jonner , I think https://github.com/micropython/micropython-lib/pull/584 has a significantly better implementation

jonnor, 2024-08-27:
Agreed! I was no aware of that merge request. Thanks :)

hellt, 2024-10-04:
Hi all,
so what is the proposed way to use imports from `typing` in micropython?

```
from typing import Union
```
raises `ImportError: no module named 'typing'`

Shall we just conditionally import typing when developing with the python interpreter and then executing in uPy?

Josverl, 2024-10-04:
My suggestions for today's versions are documented here
https://micropython-stubs.readthedocs.io/en/main/typing_mpy.html

Also there are PRs to add s typing module to either micropython, or micropython-lib