micropython/micropython-lib #192: umqtt.robust: check_msg() blocks after reconnect

puuu, 2017-06-29:
By testing the issue described in #186 I found that `check_msg()` did not work as expected in case of reconnection. This isssue may be related to #102 ...
I can reproduce it with the ESP8266 and unix port. Broker is mosquitto 1.3.4 with `persistence true`.

```python
import utime as time
from umqtt.robust import MQTTClient

def sub_cb(topic, msg):
    print((topic, msg))

c = MQTTClient("umqtt_client", "10.32.225.168")
c.DEBUG = True
c.set_callback(sub_cb)

c.connect(clean_session=False)
c.subscribe(b"foo_topic")
```

Now you can call `c.check_msg()` multiple times and see the non-blocking behaviour and if you publish a message then `sub_cb()` is called, as expected.

But if you restart the broker and call `c.check_msg()` again, you get:
```
>>> c.check_msg()
mqtt: OSError(-1,)
```
and `check_msg()` blocks until a new message is published.

This behaviour is obviously, because during reconnect, `connect()` creates a new socket and do not set non-blocking mode.
IMHO blocking during reconnection is fine, since `delay()` can be overriden, but `check_msg()` should not block, without pending message.

Turutupa, 2020-02-27:
You found a solution? Is it fixed? What is the solution to a robust connection??

EddieParis, 2020-05-05:
can you try to pull #385 and see if it fixes your issue ?

dpgeorge, 2022-10-04:
Fixed by 4dc2d5e17f1dfa0ca8af731cb1b6eec437731b25

closed 2022-10-04 by dpgeorge