micropython/micropython-lib #197: uMQTT Unsubscribe method does't exist

Bernardomrf, 2017-07-12:
Has anyone ever implemented an unsubscribe method for umqtt? Cause it would help me a lot for my master's thesis. I've tried to do that but because of my limited time I wasn't successful. 
Thanks in advance :)

edgexie, 2024-03-05:
> Has anyone ever implemented an unsubscribe method for umqtt? Cause it would help me a lot for my master's thesis. I've tried to do that but because of my limited time I wasn't successful. Thanks in advance :)

any ideas now？

Maybe we need to contribute this method.

jonnor, 2024-08-25:
Contributions are welcomed :)

mattytrentini, 2024-08-25:
There is also @peterhinch's MQTT library that is very full-featured, reliable and widely-used. And has an [`unsubscribe` method](https://github.com/peterhinch/micropython-mqtt/blob/master/README.md#324-unsubscribe). 😄

dpgeorge, 2025-12-02:
Unsubscribe added in 0d838c3a866138ab04