micropython/micropython-lib #206: urequests with ussl causes an OSError

Ahmedkalnaggar, 2017-08-19:
When trying to get or post (to https links) every n amout of seconds using the urequests library it raises an `OSError -1` . And The Error points me to `line 56` in the urequests code which is:
`s = ussl.wrap_socket(s, server_hostname=host)`
Normally I would catch any exception that occurs and try to POST it again or POST another packet.
However, if an OSError occurs all the next Requests fails too which forces me to reset the board.

dpgeorge, 2017-09-06:
Sorry for the slow response on this... which board and version of MicroPython are you using?  It may be that the device is running out of memory and can't create an SSL socket.

Ahmedkalnaggar, 2017-09-09:
@dpgeorge I am working on WiPy 2.0 Board, I think you are right. Friend advised me to use 'gc.collect()' but I don't know when should I use it. Could you confirm that might help ? or there is another better way ?

dpgeorge, 2017-09-10:
If you're using a WiPy 2.0 then I suggest you report the issue at https://forum.pycom.io/ because the bug could be in their implementation of the ussl module (urequests in know to work well on other MicroPython platforms).

pfalcon, 2017-11-02:
Closing then.

closed 2017-11-02 by pfalcon