micropython/micropython-lib #214: Approach to handling UDP in uasyncio

pfalcon, 2017-10-07:
Upstream asyncio doesn't offer "high level" support for datagram connections, along the lines of StreamReader/StreamWriter: https://docs.python.org/3/library/asyncio-stream.html . Instead lower-level support is available: https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.AbstractEventLoop.create_datagram_endpoint , based on protocol/transports paradigm, which isn't part of uasyncio.

This raises a question what to do about UDP/datagram support in uasyncio. It might even work somehow now, I know that @peterhinch used it. Like, StreamReader.read() should be read a datagram packet, and likewise StreamWriter.awrite() should be able to send it, even though both were designed with stream connection in mind. That's still a bit hacky and might not work for various edge cases too. And there're still datagram-specific calls like recvfrom() and sendto().

Two obvious approaches are:

1. Stuff more functions into StreamReader and StreamWriter, which then become misnomers, as they aren't intended for just streams any longer.
2. Introduce separate class(es) for datagrams, then also duplicate open_connection()/start_server() calls.

Even these 2 choices keep the line of deviating from the upstream, then maybe also go even further:

3. Somehow hack the API to avoid duplication of p.2.

pfalcon, 2017-10-07:
@peterhinch, @dpgeorge: Comments are welcome. (No big hurry, just wanted to have this recorded.)

dpgeorge, 2017-10-10:
Do you have any particular UDP application in mind?  For semi-related ideas about secure UDP connections see https://github.com/micropython/micropython-esp32/issues/188

pfalcon, 2017-10-10:
Well, it all started with this post by @peterhinch : https://forum.micropython.org/viewtopic.php?f=2&t=3689 . He asks how to get better working getaddrinfo() for uasyncio apps, I suggest that DNS resolver needs to written specifically for it, as a proof of concept he produces a UDP app which always waits for 1s and shows that we lack API specific for UDP. Both this and #215 stem from this.

peterhinch, 2017-10-11:
An asynchronous getaddrinfo() would have obvious application. My intended use case was an asynchronous ntptime.py (which uses getaddrinfo()).

I don't have anything to add on the API other than to agree with @pfalcon that adding functions to StreamReader and StreamWriter seems inappropriate.

pfalcon, 2017-12-15:
Finalized API merged in https://github.com/pfalcon/micropython-lib/commit/916e15ed3559e7dfd52a36f59c4dea7110acdfab .

closed 2017-12-15 by pfalcon