micropython/micropython-lib #218: OSError: [Errno 103] ECONNABORTED

Javier96Barrera, 2017-10-11:
Hi, I'm quite new to micropython. I have flashed the 1.9.2 version on my Lolin v3 (Nodemcu 12e).
I load the code using ampy command, then open terminal with Putty, and reset the board. i get this error while trying to use the sentence "connect()":

Traceback (most recent call last):
  File "main.py", line 44, in <module>
  File "umqtt/simple.py", line 56, in connect
OSError: [Errno 103] ECONNABORTED

MicroPython v1.9.2-8-gbf8f45cf on 2017-08-23; ESP module with ESP8266

`import time
import machine
import network
import gc
import dht
from umqtt.robust import MQTTClient

# DHT PIN
d = dht.DHT11(machine.Pin(14))
led = machine.Pin(16, machine.Pin.OUT)
gc.enable()

def do_connect():
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    if not wlan.isconnected():
        print('connecting to network...')
        wlan.connect('SSID', 'PASSWORD')
        while not wlan.isconnected():
            pass
    print('network config:', wlan.ifconfig())

for i in range(20):
    do_connect()
    thingspeakChannelId = "MyChannelID"  # Thingspeak Channel ID
    thingspeakChannelWriteapi = "MyWriteKey"  # Write API Key
    myMqttClient = "esp8266_mp"  
    thingspeakIoUrl = "mqtt.thingspeak.com"
    publishPeriodInSec = 15
    print("Control 1")
    c = MQTTClient(myMqttClient, thingspeakIoUrl, 1883)
    print("Control 2")
    d.measure()
    temp = d.temperature()
    humidity = d.humidity()
    print("Control 3")
    credentials = "channels/{:s}/publish/{:s}".format(thingspeakChannelId, thingspeakChannelWriteapi)
    print("Control 4")
    payload = "field1={:.1f}&field2={:.1f}\n".format(temp, humidity)
    print("Control 5")
    c.connect() #HERE
    print("Control 6")
    c.publish(credentials, payload)
    print("Control 7")
    for i in range(5):
        led.low()
        time.sleep(0.05)
        led.high()
        time.sleep(0.05)
    print("Sending data...", gc.mem_free())
    time.sleep(publishPeriodInSec)

    c.disconnect()
machine.reset()`

Thanks for your time, if there is any problem regarding the information or post structure, tell me and I will try to fix it.

dpgeorge, 2017-10-13:
This github issue tracker is mainly for bug reports and code improvements.  The report here is more of a problem using the libraries, and help in this area is best asked at https://forum.micropython.org.  So please try to ask your question there (or maybe someone had the same issue and already fixed it).

Otherwise, the general thing to try is to verify you have internet connectivity, eg:
```python
>>> import socket
>>> socket.getaddrinfo('micropython.org', 80)
```

Javier96Barrera, 2017-10-13:
Thank you for the reply, I will move this question right now. PS: I checked and yes, i got internet connectivity. Reply: [(2, 1, 0, '', ('176.58.119.26', 80))]

closed 2017-10-13 by Javier96Barrera