micropython/micropython-lib #219: wait_msg() is not working properly with Subscribe

kkumar326, 2017-10-17:
Publishing is working fine on same MQTT connection while subscribe is not working.
I traced back the error to this function: 

> wait_msg()

`res = self.sock.read(1)` is receiving: `b''` on sending data on subscribed topic.

dpgeorge, 2017-10-17:
To help fix this issue you will need to please provide more detailed information about your set-up.  Ideally provide a minimal test case that reproduces the issue, along with the version of the library you used and the version and port of MicroPython.

kkumar326, 2017-10-17:
Library is **umqtt.simple** and umqtt version is the latest one taken directly from micropython-lib.

Micropython version:

> MicroPython v1.9.2-8-gbf8f45cf on 2017-08-23; ESP module with ESP8266

Broker that I'm using is 

> Mosquitto v3.1.1

Code that I'm running is like this:

```
from umqtt.simple import MQTTClient

def switch_message(topic, msg):
	print("Remote Switch Func called")
	time.sleep(5)

c = MQTTClient(client_id = 'client-id', server='iot.xxxxx.com', port=1883, user = "username", password = "password")
c.set_callback(switch_message)
c.connect()
print("MQTT connected")

c.subscribe("client/{}/{}/switch".format("sub_p1", "sub_p2"), qos=1)
while True:
	c.check_msg()
	time.sleep(1)
```

Publishing is working fine on this exactly same connection and topic, but subscription is returning `OSError -1`. When I tried to print `self.sock.read(1)`, it just showed `b""` after publishing on that topic.

Full Error:
```
Traceback (most recent call last):
  File "main.py", line 18, in <module>
  File "umqtt/simple.py", line 202, in check_msg
  File "umqtt/simple.py", line 171, in wait_msg
OSError: -1
```

I'm also running paho mqtt subscriber on my web server which is subscribed to this topic and it is receiving everything fine.

kkumar326, 2017-10-17:
Please use this working code to test:

```
from umqtt.simple import MQTTClient

def switch_message(topic, msg):
	print("Remote Switch Func called")
	time.sleep(5)

c = MQTTClient(client_id = 'client-id-test', server='iot.airize.com', port=1883, user = "test_airize", password = "password")
c.set_callback(switch_message)
c.connect()
print("MQTT connected")

c.subscribe("client/{}/{}/switch".format("sub_p1", "sub_p2"), qos=1)
while True:
	c.check_msg()
	time.sleep(1)

```

kkumar326, 2017-10-17:
It's working fine. I made a mistake in acl file. I'm really sorry for this unnecessary issue.

closed 2017-10-17 by kkumar326

dpgeorge, 2017-10-17:
No problem.  Good that you managed to solve it yourself.