micropython/micropython-lib #220: Back references in re.py

thabubble, 2017-10-20:
Hello. I wrote something to make back references(\0-\9) work with re.sub. Not the prettiest but it does seem to work. Unfortunately, I did not understand how to commit this properly. So I'm leaving it here for now. Sorry for the inconvenience.

thabubble, 2017-10-20:
I seem to have some problems with github in my browser. I can't edit or upload files, etc.

--- .micropython/lib/re.py.bak
+++ .micropython/lib/re.py
@@ -85,7 +85,21 @@
 
     def sub(self, repl, s, count=0):
         if not callable(repl):
-            assert "\\" not in repl, "Backrefs not implemented"
+            replb = []
+            if '\\' in repl:
+                n = len(repl) - 1
+                l = f = 0
+                while True:
+                    f = repl.find('\\', f)
+                    if f == -1 or f == n:
+                        break
+                    c = repl[f+1]
+                    if c.isdigit():
+                        replb.append((int(c), l, f))
+                        l = f = f + 2
+                    else:
+                        f += 2
+                replb.append((None, l, None))
         res = ""
         while s:
             m = self.search(s)
@@ -95,6 +109,11 @@
             res += s[:beg]
             if callable(repl):
                 res += repl(m)
+            elif replb:
+                for g in replb:
+                    res += repl[g[1]:g[2]]
+                    if g[0] is not None:
+                        res += m.group(g[0])
             else:
                 res += repl
             s = s[end:]

xrmx, 2017-11-11:
@thabubble you have to fork the repo, commit your changes locally with git (while at it you want to add tests for the featured you added) and then push them to your repo on github. Then you need to open a pull request to this repo.