micropython/micropython-lib #223: Importing modules installed with upip

adamdyga, 2017-10-26:
I've build unix port of micropython (on OS.X Sierra). 
I wanted to try the 'http' module. After installing it using upip, it turned out that its dependency, the 'requests' module, is missing. Unfortunately after installing the 'request' module, the issue remains. In CLI it looks like this:

```
# install http
$ ./micropython -m upip install http
Installing to: /Users/i307115/.micropython/lib/
Warning: pypi.python.org SSL certificate is not validated
Installing http 0.02 from https://pypi.python.org/packages/e3/91/a9260805e532e33df273b8f7dffad5c51693f8f9ba5f86bedcf42a7f22eb/http-0.02.tar.gz 

# try import
$ ./micropython -c 'import http'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/i307115/.micropython/lib/http/__init__.py", line 17, in <module>
ImportError: no module named 'request'
 
# install request
$ ./micropython -m upip install request
Installing to: /Users/i307115/.micropython/lib/
Warning: pypi.python.org SSL certificate is not validated
Installing request 0.0.13 from https://pypi.python.org/packages/b0/78/bb35795b37102c0c3353cd1a24870eb92b127f6d2f982bf1b7c768e54ef1/request-0.0.13.tar.gz
 
# still the same issue :(
$ ./micropython -c 'import http'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/i307115/.micropython/lib/http/__init__.py", line 17, in <module>
ImportError: no module named 'request'
```

What am I missing?

pfalcon, 2017-10-26:
Please be sure to read the README at https://github.com/micropython/micropython-lib carefully. You install random packages not designed for MicroPython. There's no guarantee that would work, and if you'd like to express desire for that to work, you'll need to contact the maintainers of those package.

adamdyga, 2017-10-26:
OK, now I get it. Sorry for silly newbie question. Thanks

closed 2017-11-02 by pfalcon