micropython/micropython-lib #233: datetime implementation

xrmx, 2017-11-11:
Opened the issue to avoid other people trying to port Cpython's pure python datetime
implementation and find out that is blocked by the following micropython issue: https://github.com/micropython/micropython/pull/3391

That specific datetime implementation needs:

- time.strptime

And cpython tests need:

- decimal.Decimal
- test.support.is_resource_enabled
- test.support.requires_IEEE_754
- test.support.run_with_tz
- random.randrange (aliasing urandom)

pfalcon, 2017-11-11:
As the quoted ticket hints, I have the patches in my tree, and will push the for review (or directly to the repo) when other things in the queue are processed.

xrmx, 2017-11-11:
Cool, thanks! is there any of the requirements that has not been developed yet?

pfalcon, 2017-11-11:
So, I pushed the baseline CPython files to master, then https://github.com/micropython/micropython-lib/pull/234 presents my current patchset on top of that.

pfalcon, 2017-11-11:
randrange() is discussed in https://github.com/micropython/micropython-lib/pull/208

pfalcon, 2017-12-28:
Merged to mainline.

closed 2017-12-28 by pfalcon