micropython/micropython-lib #241: uasyncio: ImportError: no module named 'uselect' on Pycom LoPy ESP32

stroobandt, 2017-12-22:
As a newbee, I am still confused about when to use the `u` prefix in module names.
I got the following error when trying out the `uasyncio` module:
```
File "/flash/lib/uasyncio/__init__.py", line 2, in <module>
ImportError: no module named 'uselect'
```
Neither was `upip` of any help:
```
$ ./micropython -m upip install uselect
Installing to: /home/serge/.micropython/lib/
Warning: pypi.python.org SSL certificate is not validated
Error installing 'uselect': Package not found, packages may be partially installed
```

stroobandt, 2017-12-22:
Importing `select` instead of `uselect` in `__init__.py` resolves that issue, but then I get stuck on `utimeq` in `core.py`:

```
File "/flash/lib/uasyncio/core.py", line 2, in <module>
ImportError: no module named 'utimeq'
MicroPython 2ac6da2 on 2017-12-18; LoPy with ESP32
```

dpgeorge, 2017-12-23:
This is really an issue for Pycom/LoPy: the LoPy is not an official MicroPython board and it's up to Pycom to provide support for uasyncio.

stroobandt, 2017-12-23:
I understand. Again sorry, I am still trying to find my way amongst the different MicroPython camps, so to say.
[The following answer](https://forum.micropython.org/posting.php?mode=quote&f=15&p=24047) was received from the `uasyncio` developer on the MicroPython forum:

> I think you've uncovered a build problem with certain platforms. uselect is provided by a C module (extmod/moduslect.c) so this needs to be specified in the makefile.
> 
> Your best recourse is to raise issues on GitHub with the platform maintainers, which I see you've done against the ESP32. I'm unfamiliar with the PyCom boards and whether their firmware will automatically pick up fixes applied to the official ESP32 port.

To be honest, it is all a bit confusing when one was not there from the beginning. More so, — correct me if I am wrong — when I read that you do [occasional work for Pycom](https://pycom.io/qa-micropython-multi-threading-garbage-collector/). Perhaps a brief text up on micropython.org explaining the differences would be helpful. It all reminds me of the times when many dialects of BASIC were popular.

Anyhow, I better stick to the limited set of micropython features the Pycom microcosmos has on offer when developing for that board and complain over there when necessary…

stroobandt, 2017-12-23:
**Me too!!!** Today I have also been wasting my time with an attempt to get the extremely useful «standard» MicroPython module `uasyncio` to work on a LoPy.

Like the other poster, I also got stuck on `utimeq` not being available in the standard library of the «Pycom» MicroPython variant. (This unfortunate situation reminds me of the many BASIC dialects of times gone by!)

I am trying to design a medical device around the LoPy. Having out-of-the-box access to an event loop mechanism like the one offered by `uasyncio` is an absolute must for any serious OEM undertaking.

Related posts:
- https://forum.pycom.io/topic/781/asyncio-and-micropython/4
- https://forum.micropython.org/viewtopic.php?f=15&t=85&p=24068#p24068
- https://github.com/micropython/micropython/issues/3510

danicampora, 2017-12-23:
Hi, I just noticed this thread. We’ll get this fixed soon. The select module should be accesible using the uselect name as well.
We’ll work on full support for asyncio as well but this might take a bit more time.

Cheers,
Daniel

stroobandt, 2017-12-23:
Thanks, that would solve the `uselect` issue!
Coincidentally, this `uselect` as well as the `utimeq` issue were both raised today by someone else [over at the Pycom forum](https://forum.pycom.io/topic/2335/timeq-and-uasyncio-on-wipy-2-0-or-3-0/5).

Damien is absolutely right of course, the `utimeq` issue pertains to the Pycom MicroPython variant. A [Pycom bug report](https://github.com/pycom/pycom-micropython-sigfox/issues/112) was opened and the `utimeq` discussion will continue [over there](https://forum.pycom.io/topic/2335/timeq-and-uasyncio-on-wipy-2-0-or-3-0/5).

nevercast, 2018-01-29:
@stroobandt I'm using a WiPy2 from Pycom, with the latest firmware built from their github. `import utimeq` and `import uselect` both seem to resolve to available modules. Do you want to give this a check on your end too?

closed 2023-08-21 by andrewleech