micropython/micropython-lib #244: AttributeError: 'generator' object has no attribute 'pend_throw'

fadushin, 2017-12-31:
As of commit 203cc489c6c02ffba9b3dd3efb71d7067cc0836f, I am getting an error in uasyncio.core (line 107) when using the uhttpd library.  I have added two print lines for ret and cb, so the error is on actually line 107 as of that commit.

FDUSHIN> ret: <IORead object at 3fff6a10>
FDUSHIN> cb: <generator object 'start_server' at 3fff77d0>
Traceback (most recent call last):
  File "main.py", line 14, in <module>
  File "neolamp/__init__.py", line 60, in run
  File "uhttpd/__init__.py", line 70, in run
  File "uhttpd/__init__.py", line 407, in run
  File "uasyncio/core.py", line 140, in run_forever
  File "uasyncio/core.py", line 109, in run_forever
AttributeError: 'generator' object has no attribute 'pend_throw'

I don't know how esoteric my usage of uasyncio.start_server is -- I am passing in a locally defined function, which is annotated with @asyncio.coroutine:

https://github.com/fadushin/esp8266/blob/master/micropython/uhttpd/uhttpd/__init__.py#L401

If you like, I can try to provide a simpler test case.  I would not be surprised if I am abusing the library in some way.

Thank you!

pfalcon, 2018-01-06:
You need the latest micropython from https://github.com/pfalcon/micropython for the latest uasyncio.

Search for more info: https://github.com/micropython/micropython-lib/search?q=pend_throw&type=Issues&utf8=%E2%9C%93

closed 2018-01-06 by pfalcon

pfalcon, 2018-01-06:
Actually, that's indeed a problem, a notice was posted on the forum: https://forum.micropython.org/viewtopic.php?f=15&t=85&p=24415#p24415