micropython/micropython-lib #245: Name Error: name 'Employee' is not defined

nikhil1983, 2018-01-02:
Hey, 
While importing module ' upip ', it gives me an error of ussl module not found. 

> >>> import upip
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "upip.py", line 102, in <module>
> ImportError: no module named 'ussl'

Can anyone tell me where i will get this ussl module ?

amotl, 2019-03-09:
Hi there,

while developing the [hiveeyes-micropython-firmware](https://github.com/hiveeyes/hiveeyes-micropython-firmware), we are currently experiencing the same problem on a recent ArchLinux system:
```
micropython -m upip install -p dist-packages -r requirements-mpy.txt
Traceback (most recent call last):
  File "upip.py", line 109, in <module>
ImportError: no module named 'ussl'
```

Do you have any idea where this might coming from and how to mitigate it?

With kind regards,
Andreas.

dhylands, 2019-03-09:
MICROPY_PY_USSL doesn't appear to be defined for any of the stm32 boards, so you'll need to add that to your board file (mpconfigboard.py)

There may be additional #defines which need to be made as well to select the SSL implementation. I see that the esp8266 port and the unix port both define MICROPY_PY_USSL = 1

jonnor, 2024-08-25:
As mentioned, one may have to enable SSL in a MicroPython firmware build. This is because of size constraints.

closed 2024-08-27 by projectgus