micropython/micropython-lib #247: urequest: Cannot hanlde HTTP-response without status text

adebaumann, 2018-01-05:
I found that urequest produces an error when the http-response starts with "HTTP/1.1 500" instead of "HTTP/1.1 500 OK", complaining about needing more than 2 values to unpack. The standard python request module handles these cases fine.

pfalcon, 2018-01-05:
Are such responses compliant to the HTTP standard?

adebaumann, 2018-01-05:
According to RFC2616, "The client is not required to examine or display the Reason- Phrase" - see https://www.w3.org/Protocols/rfc2616/rfc2616-sec6.html.

From RFC 7230 (Hypertext Transfer Protocol (HTTP/1.1): Message Syntax and Routing):

    The reason-phrase element exists for the sole purpose of providing a
    textual description associated with the numeric status code, mostly
    out of deference to earlier Internet application protocols that were
    more frequently used with interactive text clients.  A client SHOULD
    ignore the reason-phrase content.

     reason-phrase  = *( HTAB / SP / VCHAR / obs-text )

I interpret the * as "optional".

pfalcon, 2018-01-06:
Thanks for references. More discussion: https://stackoverflow.com/a/17517114/496009 . So, it's still no conclusive whether `HTTP/1.0 200` is ok by the standard. What software does send out replies like that?

adebaumann, 2018-01-06:
That's "Thingsboard" (https://thingsboard.io/)

pfalcon, 2018-01-06:
Ok, thanks, let's consider it decided - I'll work on a fix for all 3 HTTP clients in micropython-lib.

pfalcon, 2018-01-14:
Got caught in with other projects... First change is went in: https://github.com/pfalcon/micropython-lib/commit/4d46561a5bdebd1352dfa1454ba582831e1da179

pfalcon, 2018-01-14:
Ok, all of `urllib.urequest`, `urequests`, `uaiohttpclient` were updated.

closed 2018-01-14 by pfalcon