micropython/micropython-lib #249: ussl: AttributeError: 'module' object has no attribute 'CERT_REQUIRED'

SergeSpinoza, 2018-01-07:
Hi! I need to connect to MQTT via SSL/TLS. 

When I try to connect to MQTT without SSL - allready is good. When I try to connect with ssl like this: 

`client = MQTTClient(CONFIG['CLIENT_ID'], CONFIG['MQTT_BROKER'], user=CONFIG['USER'], password=CONFIG['PASSWORD'], port=CONFIG['PORT'], ssl = True, ssl_params = {"cert_reqs":ussl.CERT_REQUIRED, "ca_certs":"/flash/cacert.pem"})`

I have a error: 

`AttributeError: 'module' object has no attribute 'CERT_REQUIRED'`

It's strange, because in official doc ([http://micropython.org/resources/docs/en/latest/wipy/library/ussl.html]) says that ussl.CERT_REQUIRED must be work, but it's not work =( 

What could be the problem?

manningt, 2018-10-09:
More info can be found in [issue 2781](https://github.com/micropython/micropython/issues/2781)

Swington, 2019-01-08:
@SergeSpinoza I think that CERT_REQUIRED is not actually in 'ussl' library, but in 'ssl' library, which would be 'micropython-ssl' for upip.
After importing 'micropython-ssl' onto esp8266 and importing CERT_REQUIRED from 'ssl' this problem disappeared for me.
Actually if you look in the code of 'ssl' module in micropython-lib repo, these static values (CERT_*)are created while importing 'ssl' module.
But even if you'll fix that the ssl will not work on esp8266, cause of another error, which the @manningt  linked to.