micropython/micropython-lib #250: umqtt cannot import MQTTClient

dehein, 2018-01-10:
Using the latest esp32 micropython build. Dowloaded the latest umqtt with these commands:
upip.install('micropython-umqtt.simple')
upip.install('micropython-umqtt.robust')

But as soon as i want to follow even the simplest of examples and try this:
    
   from umqtt.simple import MQTTClient
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    ImportError: cannot import name MQTTClient

but if i put the umqtt.simple lib just onto the main memomry of the esp32

    ampy --port COM6 put C:\Users\xxxx\Downloads\simple.py

and use simple.py import MQTTClient  it works.
so somehow something seems broken.

oeichler, 2018-01-28:
I had exactly the same problem.
Digging around a little bit I found out that `lib/umqtt/simple.py` was simply empty:
```
f = open('lib/umqtt/simple.py')
f.read()
```

Probably the installation silently failed in any strange way...
Repeating `upip.install('micropython-umqtt.simple')` fixed the problem.

dmurphydrtc, 2018-06-01:
Came across this issue today while using f/w esp32-20180511-v1.9.4.bin.
my simple.py file in lib/umqtt was empty. I downloaded another copy of simple.py and used ampy put to
to re-install it - ampy --port COM7 put simple.py 'lib/umqtt/simple.py'.

I went through the process of reflashing the device and reinstalling mqtt client as below but still found
the contents of simple.py were empty. Very strange.
 

"import network
import upip

sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.connect("Wifi SSID", "Wifi password")

upip.install('micropython-umqtt.simple')
upip.install('micropython-umqtt.robust')"

Mack13Podwin, 2018-07-27:
I am Using NodeMCU esp board with newest version of micropython firmware and my solution to this problem is to load `simple.py` with WEBREPL and then use 
```python
uos.mkdir()
```
and 
```python
uos.rename()
```
to put this file to `lib/umqtt`.

closed 2023-03-10 by DonPavlov