micropython/micropython-lib #257: getaddrinfo function takes 2 arguments but 4 were given

dcontris, 2018-02-05:
After installing the new version of picoweb and uasyncio, I noticed an error when calling app.run(). The error occurs because uasyncio __init__.py file contains a call to socket.getaddrinfo() with 4 arguments, when socket requires only 2. The error causes app.run() to fail.

pfalcon, 2018-02-05:
I can't reproduce this.

dcontris, 2018-02-05:
I noticed that usocket.getaddrinfo only takes 2 arguments, but uasyncio
__init__.py file passes 4.  Do u have the wrong version of usocket?  I
can't find one that use 4 arguments.

I can confirm removing the second two arguments in the initial file worked

dcontris, 2018-02-05:
Sorry, I meant to say "do I have the wrong version of usocket?"

pfalcon, 2018-02-05:
> Sorry, I meant to say "do I have the wrong version of usocket?"

Possibly, you would need to tell what version of MicroPython on what hardware/OS you use.

dcontris, 2018-02-05:
Micropython for ESP32 with the 2-5 version binaries

pfalcon, 2018-02-05:
> Micropython for ESP32

See https://github.com/pfalcon/picoweb/issues/31

dcontris, 2018-02-05:
> See pfalcon/picoweb#31 <https://github.com/pfalcon/picoweb/issues/31>

Thanks!

Sunrise17, 2018-02-13:
I have faced with the same error while trying to run `    loop.create_task(asyncio.start_server(handle_client, ipAddress, 8080))
` 
Is there some solution about it?

    File "/lib/uasyncio/__init__.py", line 235, in start_server
    TypeError: function takes 2 positional arguments but 4 were given

Sunrise17, 2018-02-13:
How i can replace current modsocket.c in esp32 with a update one in the link below?;
[https://github.com/pfalcon/micropython/blob/96ee930918079fc63729bb35a1d211c201d054ba/ports/esp32/modsocket.c](url)

dpgeorge, 2018-02-21:
This should now be fixed by https://github.com/micropython/micropython/commit/cced43feb8ffee0791f82c0358265329347f88c7

closed 2018-02-21 by dpgeorge

William04A, 2019-06-17:
Same error with request = urequests.post(requesturl, headers={"User-Agent": "Python/SomeUserAgent"})
Function takes only 2 positional arguments but four were given...

Full MicroPython traceback:
Traceback (most recent call last):
  File "<stdin>", line 34, in <module>
  File "requests.py", line 115, in post
  File "requests.py", line 53, in request
TypeError: function takes 2 positional arguments but 4 were given