micropython/micropython-lib #258: sendto() only needs two args

tschudin, 2018-02-05:
This patch made UDP sending working for me:

```
--- micropython-uasyncio.udp-0.1/uasyncio/udp.py 2018-02-04 20:52:09.000000000 +0100
+++ micropython-uasyncio.udp-0.1/uasyncio/udp.py~ 2017-12-15 23:40:03.000000000 +0100
@@ -45,7 +45,7 @@
 
 def sendto(s, buf, addr=None):
     while 1:
-        res = s.sendto(buf, addr)
+        res = s.sendto(buf, 0, addr)
         #print("send res:", res)
         if res == len(buf):
             return
```

pfalcon, 2018-02-05:
And for me it works without patching. Do you have an idea what's wrong?

Sunrise17, 2018-02-18:
File "/lib/uasyncio/udp.py", line 48, in sendto
    TypeError: function takes 3 positional arguments but 4 were given

I am receiving following error from sendto function, is there some solution about it?

jonnor, 2024-08-25:
I believe this was fixed in https://github.com/micropython/micropython-lib/commit/8d75a364314e8b4d589fefa56abb8546d69ecd0c

closed 2024-08-27 by projectgus