micropython/micropython-lib #260: Uasyncio-Udp "sendto" method error (3 positional arguments but 4 were given)

Sunrise17, 2018-02-18:
Dear @dpgeorge, I am receiving following error from "sendto" function in udp.py module, is there some solution about it?
My method as follows;
    addr = usocket.getaddrinfo("127.0.0.1", 53)[0][-1]
    yield from uasyncio.udp.sendto(sock, b"!eeeeeee", addr)

      File "/lib/uasyncio/udp.py", line 48, in sendto
    TypeError: function takes 3 positional arguments but 4 were given

dpgeorge, 2018-02-21:
It looks like uasyncio.udp was written to only work on the unix port, which allows `sendto()` to take an extra flags parameter.  As a short-term fix you can edit line 48 of udp.py and remove the "0" to make it read: `res = s.sendto(buf, addr)`.

Sunrise17, 2018-02-21:
thanks for your guidance @dpgeorge , for sendto method, i have directly used socket.sendto("message", ("127.0.0.1", port)) method in the main.code (instead of importing from udp.py), it works async. so well. Additionally, i will also apply your recommendation inside of udp.py.

(Except of this topic, i have three questions below,
1-) How can i implement OTA firmware update in micropython?
2-) During compiling of micropython, "Make" process is producing three build images as: "bootloader.bin", "partitions.bin" and "application.bin". Which one substitutes of "esp32-20180221-v1.9.3-xxx-xxxxxx.bin" in micropython download page (i mean as a structure)? For ex. Can i only use application.bin to flash board with using esptool?
3-) Micropython can be used in Manufacturing process? If yes, how can i restrict to access py files inside board?

jonnor, 2024-08-25:
I believe this was fixed in https://github.com/micropython/micropython-lib/commit/8d75a364314e8b4d589fefa56abb8546d69ecd0c

closed 2024-08-27 by projectgus