micropython/micropython-lib #267: not able to build firmware for ESP8266

dubaleeiro, 2018-04-03:
Dear all,
sorry if this is a silly mistake from my side, but after adding the content of following foldes I cannot proceed with the firmware build:
collections.defaultdict
collections.deque
collections
uasyncio.core
uasyncio.queues
uasyncio.synchro	
uasyncio.udp	
uasyncio.websocket.server	
uasyncio	

As you might have guessed, I am trying to include in the build the asyncronous functionalities. Unfortunatey I got this error message (this is only part of it actually):

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
CC build/frozen_mpy.c
build/frozen_mpy.c:1614:44: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const byte bytecode_data_collections.defaultdict_collections_defaultdict__lt_module_gt__defaultdict___new__[42] = {
                                            ^
build/frozen_mpy.c:1633:55: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const mp_rom_obj_t const_table_data_collections.defaultdict_collections_defaultdict__lt_module_gt__defaultdict___new__[2] = {
                                                       ^
build/frozen_mpy.c:1637:48: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const mp_raw_code_t raw_code_collections.defaultdict_collections_defaultdict__lt_module_gt__defaultdict___new__ = {
                                                ^
build/frozen_mpy.c:1653:44: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const byte bytecode_data_collections.defaultdict_collections_defaultdict__lt_module_gt__defaultdict___init__[29] = {
                                            ^
build/frozen_mpy.c:1667:55: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const mp_rom_obj_t const_table_data_collections.defaultdict_collections_defaultdict__lt_module_gt__defaultdict___init__[2] = {
                                                       ^
build/frozen_mpy.c:1671:48: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const mp_raw_code_t raw_code_collections.defaultdict_collections_defaultdict__lt_module_gt__defaultdict___init__ = {
                                                ^
build/frozen_mpy.c:1687:44: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const byte bytecode_data_collections.defaultdict_collections_defaultdict__lt_module_gt__defaultdict___getitem__[67] = {

...and it continues...

can anyone please help me on that issue?
Thanks a lot !

dpgeorge, 2018-05-02:
What is the directory/file structure of your modules/ directory, can you list it here?  You should put packages in subdirectories, not make files/dirs with "." in them.

dubaleeiro, 2018-05-19:
that was exactly the problem ! It worked when I created a directory `uasyncio` under `micropython/ports/esp8266/modules/` with the following files:
asyn.py
core.py
__init__.py
queues.py
synchro.py
Thanks a lot !