micropython/micropython-lib #268: usayncio: TypeError: exceptions must derive from BaseException

larsks, 2018-04-04:
I have written some code that uses the `asyncio` module.  The entrypoint point to my code looks like this:

```python
try:
    control.main()
except KeyboardInterrupt:
    pass
except Exception as exc:
    sys.print_exception(exc)
    print('* resetting in 10 seconds')
    time.sleep(10)
    machine.reset()
```

And `control.main` looks like:

```python
def main():
    [...]

    try:
        loop.run_forever()
        loop.close()
    finally:
        for task in tasklist:
            task.close()
```

I just upgraded to the version in `micropython-lib` as of 9e21d6e, and now when I interrupt my code with ^C I see the following error:

```
Traceback (most recent call last):
  File "snakecharmer/main.py", line 10, in <module>
  File "snakecharmer/control.py", line 52, in main
  File "uasyncio/__init__.py", line 246, in start_server
TypeError: exceptions must derive from BaseException
```

That seems to correspond to the following line in `uasyncio/__init__.py`:

    yield IORead(s)

What's going on here?

devbis, 2018-12-28:
Run into the same issue on the lastest micropython-lib 
Had to catch this weird exception to run my code

jonnor, 2024-08-25:
Is this still a problem with the latest MicroPython ?