micropython/micropython-lib #269: Request for package: micropython-enum

desowin, 2018-04-04:
I would like micropython to have support for `IntEnum` class. Base `Enum` would be nice to have, although just bare `IntEnum` would be enough in my opinion.

stlehmann, 2018-04-04:
Enums are especially useful when it comes to static type-checking. So it would be great if the implementation of this enum type allowed type-checking via mypy. This means all values need to be an instance of the defined enum-type. 

    >>> class MyEnum(enum.IntEnum):
    ...    A = 1
    ...    B = 2

    >>> isinstance(MyEnum.A, MyEnum)
    True

jaecktec, 2019-09-04:
I have an example where it would be awensome to have enum: 

this is a snippet from my adxl345 library:

```
class DataFormat(enum.Enum):
    """
    The DATA_FORMAT register controls the presentation of data
    to Register 0x32 through Register 0x37. All data, except that for
    the ±16 g range, must be clipped to avoid rollover.
    """
    REG = 0x31
    """Register Address"""
    SELF_TEST = 1 << 7
    """
    A setting of 1 in the SELF_TEST bit applies a self-test force to the sensor,
    causing a shift in the output data.
    A value of 0 disables the self-test force.
    """
    SPI = 1 << 6
    """
    A value of 1 in the SPI bit sets the device to 3-wire SPI mode,
    and a value of 0 sets the device to 4-wire SPI mode.
    """
    INVERT = 1 << 5
    """
    # A value of 0 in the INT_INVERT bit sets the interrupts to active
    # high, and a value of 1 sets the interrupts to active low.
    """
    FULL_RES = 1 << 3
    """
    # When this bit is set to a value of 1, the device is in full resolution
    # mode, where the output resolution increases with the g range
    # set by the range bits to maintain a 4 mg/LSB scale factor. When
    # the FULL_RES bit is set to 0, the device is in 10-bit mode, and
    # the range bits determine the maximum g range and scale factor.
    """
    JUSTIFY = 1 << 2
    """
    # A setting of 1 in the justify bit selects left-justified (MSB) mode,
    # and a setting of 0 selects right-justified mode with sign extension
    """
    RANGE_16_G = 0x03
    """Range for the adx to capture - 16g"""
    RANGE_8_G = 0x02
    """Range for the adx to capture - 8g"""
    RANGE_4_G = 0x01
    """Range for the adx to capture - 4g"""
    RANGE_2_G = 0x00
    """Range for the adx to capture - 2g"""
```

```
class ADXL345:

    def __i2c_write(self, register: enum, *bits: enum):
        def do_or(x1: enum, x2: enum): return x1.value | x2.value

        self.i2c.writeto_mem(self.addr, register.value, bytearray([
            reduce(do_or, bits)
        ]))

    def __init__(self, i2c, addr=device):
        self.addr = addr
        self.i2c = i2c
        self.__value_buffer = bytearray(6)
        self.__i2c_write(PowerCtrl.REG, PowerCtrl.MEASURE)
        self.__i2c_write(BwRate.REG, BwRate.RATE_3200_HZ)
        self.__i2c_write(DataFormat.REG, DataFormat.FULL_RES, DataFormat.RANGE_16_G)
```

Enums really make it comfortable to structure code.

roipoussiere, 2022-02-20:
I'm also interested by enum in MicroPython.

In the meantime, I found [here](https://stackoverflow.com/questions/36932/how-can-i-represent-an-enum-in-python) this workaround:

```py
def enum(**enums: int):
    return type('Enum', (), enums)

Number = enum(ONE=1, TWO=2, THREE=3)

numbers = (Number.ONE, Number.TWO)
```

matejcik, 2022-02-20:
for type-checking, I'm using the following trick:
```python
if TYPE_CHECKING:
    from enum import IntEnum
else:
    IntEnum = object

class Fruit(IntEnum):
    APPLE = 1
    PEAR = 2

def print_fruit(fruit: Fruit):
    if fruit == Fruit.APPLE:
         print("apple")

print_fruit(Fruit.APPLE)
```

brotherdust, 2022-10-14:
> 

This workaround is excellent! I combined it with const for my purposes:
```
OP_RW = enum(
        READ = const(0b1),
        WRITE = const(0b1)
    )
```

esologic, 2022-12-11:
+1 for a real implementation of this

i33l, 2023-05-26:
```python
class State(int):
    pass

State.OFF = State(0)
State.ON = State(1)

```

This approach is type-checking friendly.
```python
def foo(state: State):
    if state == State.ON:
        pass
```

andrewleech, 2023-05-26:
> +1 for a real implementation of this

The cpython implementation is surprisingly _very_ complicated and relies heavily on metaclasses which aren't supported in micropython:

https://github.com/python/cpython/blob/f585ed19ad00f78ed99ba44be5e333c056076160/Lib/enum.py#L1051

andrewleech, 2023-05-26:
> ```python
> class State(int):
>     pass
> 
> State.OFF = State(0)
> State.ON = State(1)
> 
> ```
> 
> But when have been imported getting "Unresolved attribute reference" warning.

There are a number of limitations when subclassing builtins in micropython, especially `int` because behind the scenes it's particularly optimised. Eg. https://docs.micropython.org/en/latest/genrst/builtin_types.html#no-int-conversion-for-int-derived-types-available

bredbord, 2024-03-07:
+1 for an implementation of this as well.

IhorNehrutsa, 2025-03-01:
docs/library/enum.rst: Add Enum class. #16842

IhorNehrutsa, 2025-03-05:
Implementation in:
[micropython-lib/python-stdlib/enum/enum.py: Add Enum class. #980](https://github.com/micropython/micropython-lib/pull/980)