micropython/micropython-lib #273: Request: params keyword argument for urequests

stroobandt, 2018-04-12:
Currently, `urequests` does not support [the `params` keyword](http://docs.python-requests.org/en/master/user/quickstart/#passing-parameters-in-urls) of the original Python `requests` module.

However, there exists a [better `urequests` implementation](https://github.com/lucien2k/wipy-urllib/blob/master/urequests.py) which does.
It would be great if that implementation would be adopted.

oxygene, 2021-07-10:
I'm just doing my first steps in esp32 programming with micropython, so please excuse if I miss something.
I wanted to use `urequests` to POST some data that uses `application/x-www-form-urlencoded`. So I need to `urlencode` my parameters. I can't seem to get it working, though. Installing `urllib.parse` via `upip` seems to work initially, but trying to actually import it in the python file results in

```
>>> from urllib.parse import urlencode
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/urllib/parse.py", line 30, in <module>
  File "/lib/re.py", line 11, in <module>
AttributeError: 'NoneType' object has no attribute 'func'
```

Any other way to get `urlencode` working or do I need to implement it myself?

SpotlightKid, 2021-07-10:
I created a cut-down version of `urlencode` for my [mrequests](https://github.com/SpotlightKid/mrequests) module:

https://github.com/SpotlightKid/mrequests/blob/25eac2977f75699985ab735a257c2b8ba9a7a2aa/urlencode.py

Here's a usage example:

https://github.com/SpotlightKid/mrequests/blob/25eac2977f75699985ab735a257c2b8ba9a7a2aa/examples/formencode.py

jonnor, 2024-08-25:
Several workaround are available here. Proposing to close this issue.

If someone wants to open a merge request that would also still be welcomed :)

closed 2024-08-27 by projectgus