micropython/micropython-lib #274: pypi redirects to https://pypi.org/pypi/%s/json and brakes upip

davidkwast, 2018-04-17:
Changing URL to "https://pypi.org/pypi/%s/json" appears to fix but I can't now yet because I don't have the build process setup yet.

A put modified upip.py inside the flash memory but my ESP8266 board do not have enough ram to run upip.install

```
>>> upip.install('micropython-urequests')
Installing to: /lib/
Warning: pypi.python.org SSL certificate is not validated
Error installing 'micropython-urequests': b'302', packages may be partially installed
```

drosenbauer, 2018-04-19:
Thank you, I thought I was going nuts. What's the process for this getting migrated into an official esp8266 build?

peterhinch, 2018-04-19:
I've updated [micropip](https://github.com/peterhinch/micropython-samples.git) which now works again, so at least there is a way to access libraries until upip is fixed.

pfalcon, 2018-04-19:
I'm personally busy with real life to look into this so far.

dpgeorge, 2018-04-20:
Yes it's broken.  pypi.python.org was turned off on 13th April 2018 and pypi.org is now the URL to use; see https://packaging.python.org/guides/migrating-to-pypi-org/

di, 2018-04-20:
I've made a PR to fix this: https://github.com/micropython/micropython-lib/pull/275 (I'm also a PyPI contributor/maintainer/administrator if that helps lend credibility to the patch).

dpgeorge, 2018-04-23:
This is now fixed in micropython-lib.

closed 2018-04-23 by dpgeorge