micropython/micropython-lib #283: pyboard: ImportError: no module named 'usocket'

dukeduck1984, 2018-05-26:
Hi,

When I tried to import uasyncio, an error was raised - ImportError: no module named 'usocket'.  I changed 'import usocket as _socket' to 'import socket as _socket', and type import uasyncio manually in REPL, it appeared to be fine.  But when I soft reboot the board,the error was again, this time it was 'no module named socket'.   I was really confused.

The firmware is Feb.11, 2018.

Thanks in advance.
Kaiyuan

dpgeorge, 2018-05-28:
What hardware/board are you using?  uasyncio does work on a PYBv1.x.

> and type import uasyncio manually in REPL, it appeared to be fine. But when I soft reboot the board,the error was again

Yes, if a module had an error importing then it is still registered as imported for subsequent attempts, until a soft reboot which clears the state.  That particular issue is known and discussed here: https://github.com/micropython/micropython/issues/967

dukeduck1984, 2018-05-29:
Hi Damien,

Thanks for the reply.

It's a pyb nano which is said to be fully compatible with the PYB.

Anyway I will try it again on an ESP8266.

Kaiyuan

closed 2023-06-05 by dpgeorge