micropython/micropython-lib #284: umqtt subscribe to topic with 23 bytes fails

DDDanny, 2018-05-31:
Hi, 
I can't subscribe to topic longer than 122 bytes. Publishing works fine!
I kindly ask you to verify and hopefully fix it
Thanks a lot
Danny

OK - working:
```
#122 bytes
mqtt.subscribe(b"/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/2") 
```
but one more character:
```
#123 bytes
mqtt.subscribe(b"/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/234567890/23") 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 169, in <module>
  File "umqtt_simple.py", line 154, in subscribe
  File "umqtt_simple.py", line 173, in wait_msg
OSError: -1
```

this might be related to #163 which also deals with 127 bytes

it might be related to https://github.com/micropython/micropython-lib/blob/f20d89c6aad9443a696561ca2a01f7ef0c8fb302/umqtt.simple/umqtt/simple.py#L148 as this is hitting the 128...

dpgeorge, 2018-06-01:
It looks like the subscribe method needs to handle variable length encoding of the payload size, in the same way that connect and publish do.

jonnor, 2024-08-25:
This appears ot have been fixed in the linked MRs/commits. Proposing to close this issue.

projectgus, 2024-08-27:
Looks like the author's PR was closed without merging, so I think this is still an issue.

mbuehler1990, 2024-11-19:
I had the same issue and @SpotlightKid fix worked for me. Sad that the PR is not merged