micropython/micropython-lib #286: umqtt.robust - AttributeError: 'module' object has no attribute 'MQTTClient'

infogeek, 2018-06-17:
I am trying to get umqtt.robust working but can't get past this error

>>> from umqtt.robust import MQTTClient
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "umqtt/robust.py", line 4, in <module>
AttributeError: 'module' object has no attribute 'MQTTClient'

from umqtt import robust - Works

>>> dir(robust)
['__class__', '__name__', 'utime', 'simple']

esp8266 - Wemos d1 mini
MicroPython  - v1.9.3-575-g23e9c3b

dpgeorge, 2018-06-18:
Do you also have simple.py in the umqtt directory, alongside robust.py?  That's what the `File "umqtt/robust.py", line 4, in` is about, it can't find `simple.MQTTClient`.

infogeek, 2018-06-18:
I assumed it simple.py was already inside the robust package on PYPI, installing both solved it.
Thanks Damien.

Btw here is something I have setup for my students for easily freezing there modules. http://infogeek.me/scripts/mpy-builder/ 

do you think developing it further would be useful for the wider micro python community?

dpgeorge, 2018-06-19:
> Btw here is something I have setup for my students for easily freezing there modules. http://infogeek.me/scripts/mpy-builder/
> 
> do you think developing it further would be useful for the wider micropython community?

Looks interesting!  There are quite a few tools out there already, and it's best to discuss them on the forum: https://forum.micropython.org

closed 2018-06-19 by dpgeorge