micropython/micropython-lib #290: AttributeError: 'NoneType' object has no attribute 'write'

Sushma310, 2018-06-22:
I am facing an issue with 'publish' function in umqtt.simple. It is showing 
    AttributeError: 'NoneType' object has no attribute 'write'

can you please tell me the solution for this?

dpgeorge, 2018-06-23:
Please provide more information to diagnose the problem: MicroPython port and hardware used, version of MicroPython, version of umqtt, and line number of the error.

But, it looks like you can only get this kind of error if there is no network available.

closed 2018-06-25 by Sushma310

wojsudol, 2021-01-01:
for the record. I had this error message when I was trying subscribe channel before connect to server.

nok:
``` python
    mqtt.subscribe(channel_url)
    mqtt.connect()
```

ok:
``` python
    mqtt.connect()
    mqtt.subscribe(channel_url)
```