micropython/micropython-lib #291: example code for ucollections deque

diginfo, 2018-06-26:
I have tried all of the syntax that I know might work but cannot get this working.

Would it be possible to provide some example code ??

```
>>> from ucollections import deque
>>> dir(deque)
['__class__', '__name__', 'append', 'popleft']
>>> deque([1,2,3,4])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function missing 1 required positional arguments
>>> deque([1,2,3,4],4)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: 
>>> 
```

dpgeorge, 2018-06-26:
The deque needs to be initialised with an empty tuple in the first arg, like this: `d = deque((), 4)`

The docs should be updated to provide this info.

diginfo, 2018-06-26:
Thanks Damien

Would this be the preferred way to create a uart like buffer ?

dpgeorge, 2018-06-26:
Documentation for deque was added in https://github.com/micropython/micropython/commit/b9ec6037edf5e6ff6f8f400d70f7351d1b0af67d

dpgeorge, 2018-06-26:
> Would this be the preferred way to create a uart like buffer

It depends how you want to use it.  If you just want to store bytes then a `bytearray` would be more efficient, but requires you to write code to push/pop bytes.  If storing more than just bytes then a deque is a good choice.

closed 2018-06-26 by dpgeorge

mattytrentini, 2024-04-08:
> Is the latest documentation correct for deque? 

By default, the docs ([deque docs here](https://docs.micropython.org/en/latest/library/collections.html#collections.deque)) cover the _very latest_ version of MicroPython - right now, that's the v1.23 preview release. Deque was [updated a few weeks ago](https://github.com/micropython/micropython/commit/7dff38fdc190d7b731fad8319d2ae8aa13fde18a) so if you're running v1.22 or earlier then it's changed. It's now much improved and _very_ close to the behaviour of CPython! 

You can still see the older docs, for example [v1.22 deque](https://docs.micropython.org/en/v1.22.0/library/collections.html).

Beormund, 2024-04-09:
Perfect. Thank you @mattytrentini .