micropython/micropython-lib #293: Urequests delay

fernandomorse, 2018-07-08:
When using the code example below:

```
import time
import network
import urequests

def setNetworkConfig(ssid, pwd):
    wlan = network.WLAN(network.STA_IF)
    if not wlan.isconnected():
        print('connecting to network...')
        wlan.active(True)
        wlan.connect(ssid, pwd)
        while not wlan.isconnected():
            pass
    print('network config:' ,wlan.ifconfig())

def resourceLastFeeds(protocol,url,keyPublic,keySecret,id,silent=False):

    # defining the api-endpoint
    API_ENDPOINT = protocol+"://"+url+"/api/device/"+keyPublic+"/resource/"+str(id)+"/feeds/last"

    #Headers
    if keySecret is not None:
        headers = {"Authorization":"Bearer "+keySecret,"Content-Type":"application/json"}
    else:
        headers = {"Content-type":"application/json"}

    start = time.ticks_ms()  # get millisecond counter
    # sending get
    response = urequests.get(url = API_ENDPOINT,  headers=headers)
    parsed = response.json()
    response.close()
    delta = time.ticks_diff(time.ticks_ms(), start)
    print("Delay (ms): ",delta)
    return

setNetworkConfig('SSID','PWD')
resourceLastFeeds('http', 'automacao-iot.com.br', 'C8DE1A17E97BB0D363FF6FE5E40EC544','0502B74AC1B075AAC292F7D7FE3C27936E2F5E1E539E7F6A122164061A1469F6', 13)
```

I registered a delay of 5 seconds, from the request to the closing of the same 
Commenting on the **_#parsed = response.json()_**  line, the delay passes to 360 ms (same for response.text or response.content)
Reading the urequests library, self._cached = self.raw.read(), is responsible for the delay.

```
os.uname()
(sysname='esp8266', nodename='esp8266', release='2.2.0-dev(9422289)', version='v1.9.4-8-ga9a3caad0 on 2018-05-11', machine='ESP module with ESP8266')

```
Is there any alternative to decrease this delay, using response.text, response.json or response.content ??
Am I doing something wrong?

dpgeorge, 2018-07-09:
> I registered a delay of 5 seconds, from the request to the closing of the same

I had a go at running your script above and it worked and gave approx 1.5 seconds for the execution of `urequests.get` and 7 seconds for `response.json`.  This was on the unix port of MicroPython as well as on a bare-metal target.

Testing the same request using CPython with the request module gives total execution time of around 1.5 seconds, a lot faster.

It looks like the problem is that urequests is not parsing and using the `content-length` header in the response.  Instead it just reads as much data as it can, waiting for the server to close the connection.  This may work but is not optimal in the case that the server takes a long time to time out and close the connection.  In the HTTP 1.0 RFC it states that the `content-length`, if supplied, should be used in preference to waiting for the connection to close: https://tools.ietf.org/html/rfc1945#section-7.2.2

See also the related issue #194

fernandomorse, 2018-07-09:
@pfalcon  @dpgeorge 

Thank you, problem fixed

fernandomorse, 2018-08-03:
@dpgeorge 

I used the header : 'Connection': 'Close'

But I noticed that from time to time the program freezes, I have no access to the terminal micropython (WebRepl).

The program gets frozen on line 32 of the urequest

**urequest.py**

```
def json(self):
        import ujson
        return ujson.loads(self.content)     line freezes
```
when running

`parsed = response.json()`

Not using response.json () or response.text, performs without error

closed 2018-08-07 by fernandomorse

leosok, 2021-04-11:
This is a right and great  solution (helped me with mailgun-api)