micropython/micropython-lib #294: MemoryError: memory allocation failed, allocating %u bytes

fernandomorse, 2018-07-14:
When you try to run file (test.py) with size of 18447 bytes, the error is generated: 

`MemoryError: memory allocation failed, allocating %u bytes`

When I split the same, it runs correctly. Is there any limitation of file size (py), or command that can bypass this situation

dpgeorge, 2018-07-15:
This doesn't sound like it is about any particular library and is a question more suited to the forum: http://forum.micropython.org

In short: if you pre-compile the .py to a .mpy using mpy-cross, or use frozen bytecode, it will help.

fernandomorse, 2018-07-15:
@dpgeorge 

thanks.

closed 2018-07-15 by fernandomorse

ARF1, 2018-11-10:
@dpgeorge I am also encountering the `allocation failed, allocating %u bytes` issue. My problem is not that the error itself is raised but the `%u`. Is this intended behaviour or should I open a separate issue for it?

dpgeorge, 2018-11-15:
> My problem is not that the error itself is raised but the `%u`. Is this intended behaviour or should I open a separate issue for it?

That is intended behaviour: if there is not enough memory to even allocate space to format the error message then it just gives you the unformatted string with the "%u" still in it.

amarildolacerda, 2019-06-04:
I split my  <code.py>  in two parts  ( two units) . It solved