micropython/micropython-lib #300: urequests error: IndexError: list index out of range

jperaltamalvar, 2018-08-09:


SpotlightKid, 2018-08-26:
If `getaddrinfo` is not working on your ESP32, there's nothing `urequests` can do. It means either your network is not configured properly or the `usocket` module is not working properly.

You did connect to a network before trying the examples, didn't you?

jperaltamalvar, 2018-08-28:
Yes, of course, I did successfully connect to the WIFI network. However, I think the strength of the signal is not good enough or it disconnects for some other reason. I have moved the ESP32 closer to the router, and implemented some retries (when it fails, I'm disconnecting and connecting again) and it seems to be working... let's see if it is still working by the end of this week.

SpotlightKid, 2018-08-28:
You could use the [status](http://docs.micropython.org/en/latest/esp8266/library/network.html#network.wlan.status), `isconnected` and `ifconfig` methods of your `network.WLAN` instance to check whether the network is available before using `urequests`.

jperaltamalvar, 2018-08-29:
Thanks for the feedback. Once fixed the connectivity issues it seems to be working fine, so nothing to do with `urequests` lib.

closed 2018-08-29 by jperaltamalvar

petrkr, 2020-05-05:
> If getaddrinfo is not working on your ESP32, there's nothing urequests can do

Of course it can do. Atleast it can throw relevant exception, so programmer on ESP32 can handle it by reconnect to network. But "List out of range" is terribly wrong for "host not found" error... It looks like error's from Microsoft, they also does not make sense at all.

stinos, 2020-05-05:
You're advocating that if a standard library function is not implemented correctly, all users have to deal with that. That's not normally how it works, with good reason, how else would we set up standards. Far fetched analogy: that would be like `1+1` not being `2` and then telling users they have to deal with it themselves :) A correct getaddrinfo either retuns an address, or raises an exception (which is actually how it's currently described in MicroPython's documentation, and which is likely why this PR exists: https://github.com/micropython/micropython/pull/5982), so in this case my opinion is SpotlightKid is correct and it's none of urequests business to deal with this.

amotl, 2020-05-05:
`OSError(-2); // name or service not known` would be perfect in this scenario. Thanks for referencing https://github.com/micropython/micropython/pull/5982 here, @stinos!

petrkr, 2020-05-05:
> You're advocating that if a standard library function is not implemented correctly, all users have to deal with that. That's not normally how it works, with good reason, how else would we set up standards. Far fetched analogy: that would be like `1+1` not being `2` and then telling users they have to deal with it themselves :) A correct getaddrinfo either retuns an address, or raises an exception (which is actually how it's currently described in MicroPython's documentation, and which is likely why this PR exists: [micropython/micropython#5982](https://github.com/micropython/micropython/pull/5982)), so in this case my opinion is SpotlightKid is correct and it's none of urequests business to deal with this.

OK, that could work if upcall function will return exception instead. Sorry my fault. I rather checking things more in mine code too and since this error being here for 2 years I did not looked for PR which is old 7 days..

Tested with it and that could work

```
>>> urequests.get("http://nix.cz")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 112, in get
  File "urequests.py", line 53, in request
OSError: -202
```

petrkr, 2020-05-05:
Anyway still some other bug:
```
>>> urequests.get("nix.cz")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 112, in get
  File "urequests.py", line 39, in request
ValueError: need more than 1 values to unpack
```
Cpython behaviour:
```requests.exceptions.MissingSchema: Invalid URL 'nix.cz': No schema supplied. Perhaps you meant http://nix.cz?```

SpotlightKid, 2020-05-05:
> ```
> >>> urequests.get("nix.cz")
> [...]
> ValueError: need more than 1 values to unpack
> ```

The positional arg to `get` etc. is not called `url` for nothing. `"nix.cz"` is *not* a URL.

Remember, these libraries are called **u**something*, where the **u** stands for *micro*. If they included checks for every possible *user* (developer) error, they wouldn't be micro anymore.

petrkr, 2020-05-05:
Yes I know. I just goes from linux `wget` command where also is "URL" but it supports without http too.. As we using urequests for command "wget" then it appeared too.. But this one I can handle in that our 'wget' command. Ok

TatoGman, 2022-03-31:
> urequests is working fine with HTTP, but not with HTTPS. I cannot even get the examples working:
> 
> `MicroPython v1.9.4-442-g17b512020 on 2018-08-07; ESP32 module with ESP32 Type "help()" for more information.
> 
> > > > import urequests
> > > > r = urequests.get('https://micropython.org/ks/test.html')
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > File "urequests.py", line 108, in get
> > > > File "urequests.py", line 54, in request
> > > > IndexError: list index out of range
> > > > r = urequests.get('https://www.google.com')
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > File "urequests.py", line 108, in get
> > > > File "urequests.py", line 54, in request
> > > > IndexError: list index out of range`
> 
> Next line in the "urequests.py" file returns an empty array: ai = usocket.getaddrinfo(host, port, 0, usocket.SOCK_STREAM)