micropython/micropython-lib #305: Wifi problem

fernandomorse, 2018-09-04:
Wifi disconnected, I get error osError 118 (socket operation), when I test the error, I make machine.reset () in the code, the ESP32 or ESP8266 reset, I get the following messages:

```
Started webrepl in normal mode
connecting to network ...
I (1214) phy: phy_version: 3960, 5211945, Jul 18 2018, 10:40:07, 0, 0
I (1214) wifi: mode: sta (24: 0a: c4: 09: 44: 90)
I (1214) wifi: STA_START
I (3624) wifi: STA_DISCONNECTED, reason: 201
no AP found
I (6034) wifi: STA_DISCONNECTED, reason: 201
no AP found
I (8444) wifi: STA_DISCONNECTED, reason: 201
no AP found
```

Wifi connected, ESP32 or ESP8266 intermittently freezes:

```
I (35064) wifi: n: 10, o: 10, ap: 255 255, sta: 10, prof: 1
I (35634) wifi: state: init -> auth (b0)
I (35684) wifi: state: auth -> assoc (0)
I (35724) wifi: state: assoc -> run (10)
I (35974) wifi: connected with IOT, channel 1
I (35974) wifi: pm start, type: 1

I (35974) network: CONNECTED
```

Any solution ????

dpgeorge, 2018-09-05:
This was recently implemented here: https://github.com/micropython/micropython/commit/385fa5180663221bbec033c54c37fb38f589579b

closed 2018-09-05 by dpgeorge

fernandomorse, 2018-09-05:
@dpgeorge 

I checked that

```
from network import WLAN
wlan = WLAN ()
wlan.status ()
wlan.isconnected ()
wlan.isconfig ()

```
result:

```
1010
True
('10 .1.2.141 ',' 255.255.255.0 ', '10 .1.2.1', '10 .1.2.1 ')

```
turn off my router

result:

```
1010
True
('0.0.0.0', '0.0.0.0', '0.0.0.0', '10 .1.2.1 ')
```

status() e ifconfig() not changed.

thanks