micropython/micropython-lib #307: Micropython-lib uu has no attribute "b2a_uu"

vvg123, 2018-09-10:
I trying to use `uu` for encode/decode some file in `Micropython` but failed due to 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "uu.py", line 87, in encode
  File "uu.py", line 82, in encode
AttributeError: 'module' object has no attribute 'b2a_uu'
```
Is that `uu` module missing out `b2a_uu` in Micropython?

By the way the code is correct as below?
`uu.encode("Ori.file", "Encoded.file", "/")`

SpotlightKid, 2018-09-10:
The only functions in the `uu` [module](https://github.com/micropython/micropython-lib/blob/master/uu/uu.py) are `encode` and `decode` (and `test`), as in the [documented API](https://docs.python.org/3/library/uu.html) of the CPython equivalent.

The `b2a_uu` function is supposed to come from the `binascii` module, but, alas, neither the micropython-lib implementation of this module, nor the builtin-in `ubinascii` module of MicroPython provides this function.

It seems that the `uu` module in micropython-lib was just copied from the CPython standard library without adapting it to MicroPython or testing whether it actually works. There are several other places in the code, which will not work with micropython-lib, e.g. use of `os.path` and `optparse`. So it is not in a usable state at the moment.

vvg123, 2018-09-10:
Yah, sad to know that though can use for encode some files inside Micropython. So I will close this issue then. Thanks of your reply.

vvg123, 2018-09-10:
Or is that other alternative to encoding inside Micropython?

closed 2018-09-10 by vvg123

SpotlightKid, 2018-09-10:
Does it need to be uuencoding? Why not just use base64 encoding (`ubinascii.b2a_base64()`)

vvg123, 2018-09-10:
Oh yah, I mess up the different of the `uu` and the `base64`. They are different things, thanks for pointing out, haha.