micropython/micropython-lib #318: textwrap module not usable with MicroPython ure

egranata, 2018-11-03:
I have a port of MicroPython which includes the ure module. I tried to bring in the textwrap module from this project, but ran into a couple of issues:

```
    wordsep_re = re.compile(
        r'(\s+|'                                  # any whitespace
        r'[^\s\w]*\w+[^0-9\W]-(?=\w+[^0-9\W])|'   # hyphenated words
        r'(?<=[\w\!\"\'\&\.\,\?])-{2,}(?=\w))')   # em-dash
```

This regular expression does not compile cleanly as ure does not support the (?= notation

```
_whitespace_only_re = re.compile('^[ \t]+$', re.MULTILINE)
_leading_whitespace_re = re.compile('(^[ \t]*)(?:[^ \t\n])', re.MULTILINE)
```

These two also fail to compile, in this case because the MULTILINE flag is missing

jimmo, 2022-09-19:
Looks like this module was written with the unix-ffi version of the `re` module in mind (which uses PCRE behind the scenes).

Now that we've made `textwrap` part of the python-stdlib section, it needs to be updated to no longer require these regexp features.

jimmo, 2022-09-20:
@brianlinuxing

What do you actually need this module for? I'm not seeing any obvious reasons why a microcontroller needs to do text wrapping.

More specifically, if we were to make a much simpler subset implementation rather than making all the features of the CPython version work, what exact features would you require?

jimmo, 2022-09-20:
@brianlinuxing

If you don't use the `dedent` function or don't require `break_on_hyphens` to be set to True for the `wrap`, then all you have to do is remove the lines that use the `re` features that aren't supported and this will work fine.