micropython/micropython-lib #319: datetime 'tzinfo' has no attribute '__new__'

rdagger, 2018-11-06:
When I try to import datetime using latest build of LoBo MicroPython I get the following error:

> >>> import datetime
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "datetime.py", line 1919, in <module>
>   File "datetime.py", line 1836, in _create
> AttributeError: type object 'tzinfo' has no attribute '__new__'
> >>>

rdagger, 2018-11-09:
I also tested with the latest Micropython.org build esp32spiram-20181109-v1.9.4-683-gd94aa577a.bin
It produces the same error message as the LoBo build.

YTBeater, 2019-06-28:
@miguelgrinberg Thanks you for your fixes. That work for me too.

stroobandt, 2019-11-13:
@miguelgrinberg Thanks for the fix Miguel. That works here too.

jonnor, 2024-08-25:
This seems to have been fixed. 
Using MicroPython 1.23 and the Unix port, I can successfully import datetime.

micropython -m mip install datetime
```python
import datetime
```

projectgus, 2024-08-27:
Yes, I think this code got rewritten in fc86070ffb8aeea.

closed 2024-08-27 by projectgus