micropython/micropython-lib #322: utarfile  error

23233, 2018-12-20:
##  erorr code 
```
f1 = tarfile.TarFile("test.zip")
<TarFile object at 3ffec980>

f1.next() 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "local_lib/utarfile.py", line 88, in __next__
  File "local_lib/utarfile.py", line 78, in next
UnicodeError: 

bytearray(b'PK\x03\x04\n\x00\x00\x00\x00\x00\x93\x98\xf0L\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x12\x00\t\x00gosignaler-master/UT\x05\x00\x01\xb7NM[PK\x03\x04\n\x00\x00\x00\x00\x00\x93\x98\xf0L\xebu\xd3\x04!\x00\x00\x00!\x00\x00\x00\x1c\x00\t\x00gosignaler-ma')

in 89 line code is 

 d.name = str(h.name, "utf-8").rstrip("\0")

```
## i want to zip file extractor  how do you doing?

dhylands, 2018-12-20:
TarFile is for extracting tar files not zip files. I'm not sure anyone had written a zip file extracted. The core compression algorithm is available in uzlib.

23233, 2018-12-21:
**@dhylands  Thank you replay my question ！**

but i'm try to extractor a tar files , error same!!

please help me 
 
i don't know how about and how to work~ 

thanks you 

hardware：esp32
firmware：1.9.1
tar files : create test.py files,write some line,  run  tar zcvf test.tar.gz test.py

dhylands, 2018-12-21:
.tar.gz is a tar file that has been compressed. Try just a plain tar file.

23233, 2018-12-22:
nice create a tar file this is working ~~

tar -cf test.tar test.py

thanks @dhylands !!!

closed 2018-12-22 by 23233