micropython/micropython-lib #329: micropython-datetime PYPI package needs updating

averes-avidid, 2019-02-08:
The micropython-datetime package at https://pypi.org/project/micropython-datetime/ still has the dummy module. Please update the package with the latest version of micropython-datetime.

gnbl, 2023-02-18:
What's the status here?

It seems there is a module:
- https://github.com/micropython/micropython-lib/tree/master/python-stdlib/datetime

But no documentation:
- https://docs.micropython.org/en/latest/search.html?q=datetime&check_keywords=yes&area=default

And it does not seem to be in the current build(s):
```
MicroPython v1.19.1-869-gde1f1dd16 on 2023-02-16; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import datetime
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
  ImportError: no module named 'datetime'
```

gnbl, 2023-02-18:
Uh, sorry, needs to be installed - see the readme: https://github.com/micropython/micropython-lib

jonnor, 2024-08-25:
Packages for MicroPython should *not* be installed from PyPi / with pip. Instead the `mip` package manager should be used (since MicroPython 1.20). See the official documentation at https://docs.micropython.org/en/latest/reference/packages.html

closed 2024-08-27 by projectgus