micropython/micropython-lib #333: "ValueError: generator already executing" as a result of exhausted heap on ESP8266

eltomek, 2019-03-03:
uasyncio encounters an issue and interrupts program execution at
`await asyncio.sleep(delay_secs)`
line with 
`ValueError: generator already executing`
when the available heap size drops below ~10500 bytes at ESP8266 (Wemos D1 mini board).

As a workaround, one can either force garbage collector to collect as often as possible (`gc.collect`) or increase GC's sensitivity by setting the `gc.threshold`'s parameter empirically.

eltomek, 2019-03-03:
Seems like it's strongly related to MQTTClient object creation, need to investiage more before logging a defect.

closed 2019-03-03 by eltomek

devbis, 2019-03-13:
Thanks for the investigation. Are you sure that this is caused by exhausted heap?
I quite regularly can see this exception, but I use deep sleep with a short working cycle. So it is quite weird that heap exhausting randomly.

eltomek, 2019-03-19:
It is likely unrelated to uasyncio itself however it very much helps to run it within uasyncio. I have performed dozens of tests to see which part of my code caused the issue and pinned this:

`            client = MQTTClient(client_id=myMqttClient, 
                                server=THINGSPEAK_URL, 
                                user=THINGSPEAK_USER_ID, 
                                password=THINGSPEAK_MQTT_API_KEY, 
                                ssl=False)

            client.connect()  # <- this part generates issues when the heap is running low, but not always. 
`
`gc.collect()` seems to be just a workaround, but at least worked for me, right now I will be running the script for 3 weeks continuously, will see if it will make it.