micropython/micropython-lib #335: Bug in uasyncio lib (or in example of use but rather not)

mvincm, 2019-03-30:
Hello,

First of all some information:

Build: MicroPython v1.10-247-g0fb15fc3f on 2019-03-29; ESP module with ESP8266
Hardware: ESP8266
Included libs: standard + uasyncio + uasyncio.core
Script: https://github.com/micropython/micropython-lib/blob/master/uasyncio/example_http_server.py

Fresh build, fresh install no other scripts or errors.

Bug:
```
<StreamReader <socket state=3 timeout=0 incoming=0 off=0> <socket state=3 timeout=0 incoming=0 off=0>> <StreamWriter <socket state=3 timeout=0 incoming=0 off=0>>
================
b'GET / HTTP/1.1\r\nContent-Type: application/x-www-form-urlencoded\r\ncache-control: no-cache\r\nPostman-Token: ccda70f3-8215-4512-9168-b66baf349ac9\r\nUser-Agent: PostmanRuntime/7.6.0\r\nAccept: */*\r\nHost: 192.168.1.13\r\naccept-encoding: gzip, deflate\r\nConnection: keep-alive\r\n\r\n'
After response write
Traceback (most recent call last):
  File "main.py", line 20, in <module>
  File "uasyncio/core.py", line 161, in run_forever
  File "uasyncio/core.py", line 136, in run_forever
  File "uasyncio/__init__.py", line 60, in remove_writer
TypeError: function takes 2 positional arguments but 3 were given

MicroPython v1.10-247-g0fb15fc3f on 2019-03-29; ESP module with ESP8266
Type "help()" for more information.
>>>
```

I would be vert grateful if someone could take a look in this issue. Maybe some one have the same?

Best regards,
MvincM

peterhinch, 2019-04-02:
This is a version problem. The version on PyPi is incompatible with the official build of firmware. With that firmware you need to use the version located [in the official library](https://github.com/micropython/micropython-lib). The issue is explained [here](https://github.com/peterhinch/micropython-samples) - follow the link.

mvincm, 2019-04-02:
Hello.

I can confirm that with the official library it is working just fine. Thanks !!!

closed 2019-04-02 by mvincm