micropython/micropython-lib #350: AttributeError: 'defaultdict' object has no attribute 'items'

Xiretza, 2019-09-09:
```pycon
>>> from collections import defaultdict
>>> dd = defaultdict(list)
>>> dd
<defaultdict object at 2006e7d0>
>>> dd.items()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'defaultdict' object has no attribute 'items'
```

`defaultdict` doesn't have an `items()` method, so iterating over it is a bit of a pain in the neck. Also no `__repr__()`, but that's not totally necessary I guess.

Rohith295, 2021-06-15:
Can I work on this issue? @xrmx

jonnor, 2024-08-25:
This seems to still be the case - no "items()" function in the defaultdict module.

Anyone is welcomed to open merge requests in the MicroPython project :)