micropython/micropython-lib #357: can't await mqtt.simple publish method

alirezaimi, 2019-10-22:
Hi
I'm using umqtt.simple and there is problem when i await mqtt.publish in my code ! 
`'NoneType' object isn't iterable`
Is there any way to solve this issue ?
Or use other lib like https://github.com/peterhinch/micropython-mqtt/blob/master/mqtt_as  ?

2niuhe, 2020-01-06:
The umqtt.simple module is sync.You'd better use peterhinch's uasync mqtt lib.

While  you only need subscribe to some topic.
my solution is :
- use check_msg() instead of wait_msg(),cause check_msg() return immediately.

 ```python
import uasyncio
from umqtt.simple import MQTTClient

loop = uasyncio.get_event_loop()
async def my_callback(topic,msg):
    await uasyncio.sleep(1)
    
def sub_callback(topic, msg):
    global loop
    loop.create_task(my_callback(topic,msg))

async def my_main():
    c = MQTTClient("umqtt_client", server)
    c.set_callback(sub_callback)
    c.connect()
    c.subscribe(b'foo_con')
    pass

loop.creat_task(my_main())
loop.run_forever()
```

alirezaimi, 2020-01-07:
@2niuhe thanks, but using umqtt_async have its own problem : https://forum.micropython.org/viewtopic.php?f=15&t=7144

thanks for your solution but i had problem in publish part not others .