micropython/micropython-lib #367: urequest IndexError

tatianagasser, 2020-01-09:
Hello, 

I have this error : 
[image]

All I notice is that the problem appears when the data is very important. Do you have any idea how to solve the problem? Thank you. 

Example of data :
{'1': {'mac': b'XXXX', 'reg1': {'timestamp': (2020, 1, 2, 14, 15, 1, 221154, None), 'data': b'XXXX'},'reg2': {'timestamp': (2020, 1, 2, 14, 15, 5, 78284, None), 'data': b'XXXX'}}, '0': {'mac': b'XXXX', 'reg1': {'timestamp': (2020, 1, 2, 14, 15, 1, 94182, None), 'data': b'XXXX'}}, '2': {'mac': b'XXXX', 'reg1': {'timestamp': (2020, 1, 2, 14, 15, 5, 78284, None), 'data': b'XXXX'},'reg2': {'timestamp': (2020, 1, 2, 14, 15, 5, 78284, None), 'data': b'XXXX'}}}

Code : 
```
def request(payload):
    print("DSN :")
    print(usocket.dnsserver())
    usocket.dnsserver(0, '8.8.8.8')
    usocket.dnsserver(1, '10.0.0.1')
    print(usocket.dnsserver())
    flag=False
    while(flag==False):
        try :
            print("Using library urequest...")
            r= urequest.post("https://labs.de/node-red",json=payload)
            print("Return:", r.text) #r.content
            flag=True
            return r
        except OSError as err:
            print("OUPS ! OS Error {0}".format(err))

def deconnect_to_LTE():
    try :
        pycom.rgbled(0xd5362a) # red
        lte.disconnect()
        print("Disconnected!")
        lte.detach()
        print("Dettached!")
        lte.deinit()
    except OSError as err:
        print('Oups..')
```

jonnor, 2024-08-25:
Hi. There have been many improvements to the `request` library over the years. Please re-test to see if this issue also exists on the latest version.

For information, there is also an ascyncio HTTP client available in `aiohttp` module.

projectgus, 2024-08-27:
This user's account has been deleted, so I think we're too late for a re-test here.

closed 2024-08-27 by projectgus