micropython/micropython-lib #369: hmac code not compatible with uhashlib implementations

tmmorin, 2020-02-03:
I noticed that micropython's `hmac` implementation [relies on the existence of `.copy()` on instances of shaXX](https://github.com/micropython/micropython-lib/blob/master/hmac/hmac.py#L118), which works fine with the implementation provided in `hashlib._sha224`/`256`/`384`/`512`, but does *not* work with the implementations that `hashlib` auto-loads from `uhashlib` when `uhashlib` is present (since [this change](https://github.com/micropython/micropython-lib/issues/235)).

I tested recent versions micropython-lib master and micropython 1.12).

```
$ cat test2.py 
import hmac
from hashlib import sha256
print(hmac.new(b'test', msg=b'test', digestmod=sha256).hexdigest())
```
```
$ micropython test2.py 
Warning: No block_size attribute on given digest object; Assuming 64.
Traceback (most recent call last):
  File "test2.py", line 4, in <module>
  File "/home/treizh/.micropython/lib/hmac.py", line 137, in hexdigest
  File "/home/treizh/.micropython/lib/hmac.py", line 120, in _current
AttributeError: 'sha256' object has no attribute 'copy'
```

Forcing the use of the implementation under `hashlib._sha256` works though:

```
$ cat test.py 
import hmac
from hashlib._sha256 import sha256
```
```
print(hmac.new(b'test', msg=b'test', digestmod=sha256).hexdigest())
$ micropython test.py 
88cd2108b5347d973cf39cdf9053d7dd42704876d8c9a9bd8e2d168259d3ddf7
```

(this workaround unfortunately does not work for SHA1 since SHA1 has no implementation in `hashlib`)

I presume that whether `hmac` should be implemented without using relying on `copy()`, or at least should not autoload `hashlib` (this last solution would leave HMAC SHA1 unimplemented).

tmmorin, 2020-02-03:
@peter-conalgo: perhaps are you interested in this issue ?

peter-conalgo, 2020-02-03:
Somewhat. I used this code as an example, and made my own when I needed HMAC recently. Pared down features I didn't need so that in the end, I didn't need to copy state.

My version:

<https://github.com/Coldcard/firmware/blob/119d7f79d82d98d35e53a941bbc065fa81182e50/shared/hmac.py>

tmmorin, 2020-02-03:
Thanks @peter-conalgo. I'll try your code !

cositotito, 2020-04-24:
hello the code error digi

jimmo, 2022-08-08:
See https://github.com/micropython/micropython-lib/pull/515 for a fix for this (sorry this has taken so long)

dpgeorge, 2022-08-08:
Fixed by f95260d7e3c123a8b1907ded095e4c25e4251085

closed 2022-08-08 by dpgeorge