micropython/micropython-lib #374: urllib.urequest.urlopen fails with mbedtls_ssl_handshake error on ESP32

czAdamV, 2020-03-14:
I'm running into problems when trying to make an HTTP request to one particular page using urllib.urequest installed by upip in micropython (version micropython/micropython@eae495a71 with russhughes/st7789_mpy@ff343dc compiled in):
```
>>> from urllib.urequest import urlopen
>>> res = urlopen('https://aladinonline.androworks.org/')
mbedtls_ssl_handshake error: -4290
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/urllib/urequest.py", line 30, in urlopen
OSError: [Errno 5] EIO
```
The WiFi connection seems to be working because I can connect to other pages just fine, I didn't find a single page other than this one that would fail with the same error. I tried looking for the `-4290` error code in mbedtls documentation, but I couldn't find anything.

The big picture here is that I'm trying to build a tiny weather station with the TTGO T-Display board. The page that I'm trying to connect here offers a very convenient API for data from a particular atmospheric model used for the weather forecast in Europe (ALADIN). My initial goal is to make a request to the API, store the response (which contains a forecast for the next two days) and show some information from it on the board's display. This is my first micropython project though and I have no clue how to debug the issue further. Any ideas?

czAdamV, 2020-03-16:
I tried looking into urlopen code, and I managed to replicate the issue using just micropython library, but I'm unsure what to do next. The following fails with the same error as the `urlopen` call, suggesting this might be a bug in micropython itself, not in micropython-lib's implementation of `urllib`:
```
>>> s = usocket.socket()
>>> s.connect(('185.8.236.214', 443))
>>> s = ussl.wrap_socket(s, server_hostname="aladinonline.androworks.org")
mbedtls_ssl_handshake error: -4290
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO
```
The micropython documentation, however, doesn't say anything about the `server_hostname` parameter of `ussl.wrap_socket` function. If I omit the parameter, the connection works fine:
```
>>> s = usocket.socket()
>>> s.connect(('185.8.236.214', 443))
>>> s = ussl.wrap_socket(s)
>>> s.write(b"GET / HTTP/1.0\r\nHost: aladinonline.androworks.org\r\n\r\n")
53
>>> s.readline()
b'HTTP/1.1 200 OK\r\n'
```
but since the parameter isn't documented, I don't know what other side effects this change has. Is this a micropython bug and should I open an issue there (and close this one), or does it belong here?

mardigras2020, 2020-03-24:
Hi - had the same issue an i downgraded the ESP Firmware to esp32-idf3-20191220-v1.12.bin 
after flashing the board with the stable release - it worked for me.

czAdamV, 2020-03-24:
Oh, you're right @mardigras2020, it does work with esp32-idf3-20191220-v1.12.bin. I didn't even realise the other one was a nightly version when I downloaded it, that's why I didn't try it with a stable one. Now it really seems like a bug in micropython itself. I'll try to find out which commit introduced it, open an issue in the other repository and close this one.

czAdamV, 2020-03-26:
Closing the issue as this seems to be a micropython bug after all.

closed 2020-03-26 by czAdamV