micropython/micropython-lib #375: Is this the only source for micropython modules?

larsks, 2020-03-16:
@dpgeorge there was a commit (1509830) a couple years ago noting that this repository is unmaintained. It was reverted a couple of days later, but other than that revert there's been no activity here for over two years. Should we just archive this repository so that people aren't tempted to file issues here anymore? The code would continue to be available in its current state.

There appears to be more activity at https://github.com/pfalcon/pycopy-lib, so maybe it's better to redirect people there (although I don't know if those modules are necessarily going to be compatible with what's here).

jimmo, 2020-03-20:
Hi @larsks -- I've raised PR #376 as a way to continue this discussion.

Agree wholeheartedly that it's confusing when people find this repo, and not just because of the lack of activity but also it's really confusing how to actually use most of these modules.

I don't really want to comment on pycopy, other than maybe to say that a good way to summarise my goal with #376 is that it clarifies the difference in direction between MicroPython and pycopy.

larsks, 2020-03-20:
The proposal in #376 looks fantastic!

closed 2020-03-20 by larsks